va3wam/TWIPe #3: RED and GREEN LEDs not working

va3wam, 2020-04-17:
They did work briefly but then stopped. Multi meter voltage reading for both LEDs are the same at ~5VDC. The thinking is that the LEDs cannot handle that much current/voltage. The WHITE and AMBER LEDs are still working but are being driven by much lower volage (~1.9VDC). Also note that the WHITE LCD may have a higher tolerance than the other LEDs.

va3wam, 2020-04-17:
Likely need to modify the PCB design and fabricate a new version which passes less voltage/current to the RED and GREEN LEDs. Other than the LEDs everything is working so this is not an urgent fix.

nerdoug, 2020-04-19:
white switch specs: https://www.digikey.ca/products/en?keywords=360-3177-ND
other switch specs: https://www.digikey.ca/product-detail/en/UB16SKG035D-DD/UB16SKG035D-DD-ND/1027349/?itemSeq=296939002

The White push button has a higher tolerance for the actual switch, because it handles 20V and 3A.
It's LED specs are close to the other colours. Specs for our LEDs are:
     White: 3.3V  15mA
     Red:  1.85V  20mA
     Green: 2.25V  20mA
     Amber: 2.0V  20mA

Applied supply voltage in SB7 with ground fix jumper, as measured on Doug's board:
     White: 1.86V
     Red:  4.97V
     Green:  5.02V
     Amber: 2.03V

To correct a supply voltage that is higher than the rated voltage of a LED, you put a ballast resistor in series with the LED. The value of the resistor is calculated as:
   R = (Vsupply - Vled)/Cled
where 
     Vsupply is the supply voltage, assumed to be higher than LED's rated voltage
     Vled is the rated voltage for the LED, given above for our LEDs
     Cled is the current rating of the LED

The ballast resisters we would need are:
    White: unneeded, because supply voltage is less than LED's voltage spec
    Red:  (4.97 - 1.85) / 0.020 = 156 Ohms
    Green: (5.02 - 2.25) / 0.020 = 138.5 Ohms
    Amber: unneeded because supply voltage is very close to LED's voltage spec.

Possible Actions:
    - put appropriate ballast resistors into design of next PCB iteration
    - measure supply voltages on Andrew's SB7 to verify consistency
    - de-solder pushbuttons from SB6 board and use them for testing and experiments
    - possibly desolder 2 switches from an SB7 board to test, and modify circuit, but it's risky.

nerdoug, 2020-04-19:
To: Andrew Mitchell <va3wam@gmail.com>

I put a bunch of technical discussion into issue #3, with these suggested actions on your part:
- measure the voltages on all 4 LEDs on your board. I had to hack my test program to keep the amber LED on solid so I could measure it.
- desolder the push buttons from the earlier SB6 board so we can test and experiment with them.
Waddaya think?

nerdoug, 2020-04-20:
Found the Official NKK ballast resistor reference:
https://www.nkkswitches.com/pdf/BallastResistors.pdf
which agrees with previous calculations. adding the resistors brought led voltage for red LED to good value, but only reduced grenn LED's voltage form 5.02 to 4.7 volts. Suspect  green LED has been damaged and is now out of spec.

I'm not going to investigate further, since boards are usable at this point, and green LED is semi functional, but very dim.

now need to make changes to board design so next iteration has correct LED voltages.

va3wam, 2020-04-21:
Change title to more accurately reflect what is not working now as the RED light on my robot came back to life. Doug has plans (see his last comment on  this thread) to address this non-urgent issue if/when we spin a new version of the PCB.

va3wam, 2020-04-22:
Doug has a plan to modify the LED circuit to put things in the proper voltage range for the RED and GREEN LEDs. Moving to request for next PCB design.