va3wam/TWIPe #100: Break up the monolithic code base into seperate pieces

va3wam, 2020-12-30:
Make use of linker to merge multiple modules.
Use named code sections to enable modularization.

nerdoug, 2021-01-11:
named code sections doesn't seem to be feasible in the C++ compiler that VSC/PIO use.
We can use libraries as a model where:
.h file has extern statements that specify format of routine calls. This removes the need to have an extern statement in every module that calls a given routine, which enforces consistency.
.cpp files contain actual routine code

Can have multiple routines per module, notably routine(s) to be called from setup, plus one or more routines to be called somewhere within loop().

va3wam, 2021-01-12:
Ok thanks Doug. I have some libraries going to show you on Wednesday. 

Andrew