va3wam/TWIPe #95: Create a lighter chasis

va3wam, 2020-12-30:
Come up with a target weight and motor torque that allow for fuller range of speeds of movement

nerdoug, 2021-01-12:
recommendation from Even Pratten: We build a lot of the robot out of Lexan that is cut on our 2D CNC machine that I can't remember the name of. If you have never worked with Lexan before, it is reasonably cheap, and usually used as bulletproof glass. It is bendable under heat or in a vice with some dedication, it can also be cut by a saw or CNC machine. I have stood on thin sheets of it before, and it can easily hold my weight. The only thing is, like most strong materials, when it breaks, it will explode apart into tiny pieces.

va3wam, 2021-01-12:
Cool. This stuff can be bought at Home Depot. I actually think that zippitwippi is made with it.

Andrew